pinout-xyz/Pinout.xyz #377: Accuracy of alternative pin uses for Raspberry Pi 4

rgov, 2020-09-14:
The Raspberry Pi 4 introduced additional UARTs which can be enabled with the dtoverlays `uart2`, `uart3`, `uart4`, and `uart5` respectively. Enabling an overlay repurposes some GPIO pins for the TX/RX.

I have had a hard time finding documentation for which pins get used but my best source is [this table](https://github.com/RPi-Distro/raspi-gpio/blob/master/raspi-gpio.c#L99) in the sources of the `raspi-gpio` tool originally written by the then-Director of Hardware Engineering of the Raspberry Pi Foundation, so I take it to be a reputable source.

The table suggests that the pinouts for the BCM2711 used by the Raspberry Pi 4 differ from those of the BCM2708 used by other models. Here is the differences for GPIO 0 through 27, highlighted orange where they are not the same:

[image]

It would be good to have this information presented on the site.

lurch, 2020-09-15:
Duplicate of #322 ?

Gadgetoid, 2020-09-15:
Either way it's clear I need to do something about this!

@rgov does the table view presented here https://github.com/Gadgetoid/Pinout.xyz/issues/322 with @lurch 's modifications work for you?

(Closing this to focus discussion on #322)

closed 2020-09-15 by Gadgetoid

rgov, 2020-09-15:
@lurch Thanks, fixed the diagram for posterity. @Gadgetoid The mockup you posted looks good, though I agree highlighting the differences would be good.