pinout-xyz/Pinout.xyz #378: Blinkt! Amend with note for early versions

nickwebcouk, 2020-09-15:
Regarding - Can a note be added to say that early versions of Blinkt! ran power for 5V via Pin 2, and not Pin 4. 

See

https://twitter.com/nickwebcouk/status/1305838819816673280?s=20

Thanks,

Nick

Gadgetoid, 2020-09-15:
Yes, yes I can. Sorry for sending you on a wild goose chase.

nickwebcouk, 2020-09-15:
Absolutely no goose chase at all! I appreciate this will probably be quite rare but if it saves some head scratching for someone in the future!

Gadgetoid, 2020-09-15:
Oh it looks like there was already a message, but it's virtually impossible to see unless you're looking for it. Trying to refactor the page to be clearer:

[image]

nickwebcouk, 2020-09-15:
Well thats embarrassing. I read that page a good few times and never even noticed. Sorry for the extra work!

Gadgetoid, 2020-09-15:
It's okay, I only saw it because I was *looking* for it (and wrote it, probably). If this results in better clarity for other users, it's time well spent!

closed 2020-09-18 by Gadgetoid