pinout-xyz/Pinout.xyz #379: "RAM: NoneMb" Possible bug?

alex-engelmann, 2020-09-26:
Hi!

I tried pinout today for the first time and was very glad it existed in case I ever lost the card :)  However, I noticed a small bug in the information it provides for my board.   I've included a screenshot where it shows my memory as `NoneMb` which is a bit strange, and I've also included some basic system information.

I didn't see an open issue around it or settings in the `man` pages.  Happy to help fix this, let me know if you'd like more information.
[image]

lurch, 2020-09-27:
The `pinout` [command-line tool](https://gpiozero.readthedocs.io/en/stable/cli_tools.html) is actually provided by GpioZero! pinout.xyz is just a website and doesn't install anything on your Pi.

Could you please close this issue and open a new issue at https://github.com/gpiozero/gpiozero/issues/ ?  Thanks.

lurch, 2020-09-27:
Ah, a bit more digging and it looks like this problem was [already fixed](https://github.com/gpiozero/gpiozero/commit/aeb9d30056ec97e6bf896152e71a870bd0099b4e), but there hasn't been a release of gpiozero with the fix included yet.
ping @bennuttall @waveform80

alex-engelmann, 2020-09-27:
Hi @lurch thanks for letting me know and looking into it.  I'll go ahead and close this issue since it's likely to fixed soon.  Cheers!

closed 2020-09-27 by alex-engelmann

bennuttall, 2020-09-27:
Yeah we should do another release really, will look into it soon.