pinout-xyz/Pinout.xyz #384: Pirate Audio: Speaker pinout changed for newer boards

richteel, 2020-10-01:
Apparently Pimoroni changed the Button Y from GPIO 20 to GPIO 24 on boards manufactured after 23 January 2020. (SOURCE: [https://github.com/pimoroni/pirate-audio/issues/16](https://github.com/pimoroni/pirate-audio/issues/16) sandyjmacdonald stated, “On new versions of the Pirate Audio boards (after 23 January 2020), the button previously connected to BCM20 is now connected to BCM24.”)

The board that I recently purchased uses GPIO Pin 24 for Button Y. I believe the current page is correct for older boards but not newer boards. The page should indicate that there is more than one version of the board and that the pinouts are different between versions.

Attached are photos of my board. This is a newer version using GPIO 24. I do not see any markings showing it as a different version/revision. I included the photos here in case someone spots something or has an older board and can tell the difference.

[image]

[image]

axelsimon, 2020-12-03:
The easiest solution here might be to create a second version of this board, basing it on [the first](https://github.com/Gadgetoid/Pinout.xyz/blob/master/src/en/overlay/pimoroni-pirate-audio-speaker.md).
Then, put a warning on each page with a link to the other board (and ideally a way to differentiate between both!).

jrobertson-insite, 2020-12-16:
I found the same. Unfortunately on my own before finding this thread. I also found I needed 3v3 power (pin 17) to get it working without the full header connected.

Gadgetoid, 2020-12-16:
I'll have to double check the schematics when I get a chance to see which boards need a 3v3 supply. I've made an attempt to clarify the pin re-assignment, though. I vaguely recall we might have only released one board  type with GPIO 20 as Button Y before realizing we'd been silly and used PCM DIN and moving to GPIO 24.

closed 2026-08-18 by Gadgetoid