pinout-xyz/Pinout.xyz #389: 4-mic-respeaker

ray-leiter, 2020-11-29:
I think the pinout diagram (https://pinout.xyz/pinout/respeaker_4_mic_array) shown for this board has some errors.
They are associated with the 4 pin JST connector the Seeed people refer to as "GPIO12" used to provide access to GPIO 12 & 13.
(They call the connector a "Grove" connector.)
The pins in question are designated as D0, D1, Vcc, Gnd.
The problem seems to stem from the lack of a clear indication, from Seeed, as to the pin numbers of the JST connector.
Looking at the edge of the respeaker board with the JST connector on the bottom of the board in your view, the pins are numbered, left to right: 1,2,3,4.
If you look at the board with the JST connector on top of the board, the the pin numbering, left to right is 4,3,2,1.
The pin assignments, in pin number order are: 1: D0, 2: D1, 3: Vcc, 4: Gnd.
D0 is GPIO12 while D1 is GPIO13.
In the diagram given by the URL above, J8 pin #32 (GPIO 12) is shown as "GP12 pin 4" and should be "GP12 pin 1".
In the diagram given by the URL above, J8 pin #33 (GPIO 13) is shown as "GP12 pin 3" and should be "GP12 pin 2".

I did not check the other 4 pin JST connector used for I2C -- it may be wrong as well!

I had to figure this out since I couldn't find anywhere on the Seeed website any diagram of the connector and its pin assignments -- they leave much to be desired in their documentation.

ray-leiter, 2020-12-02:
Its been 3 days. I don't think anybody really cares if the pinout diagrams are wrong.
I have since determined the I2C connector pinouts are assigned in the same order as the GP12 pins.
For some reason, whoever did the pinout diagram for the ReSpeaker 4 Mic Array didn't see fit to include the pin assignments for the I2C connector. They did show the GPIO 2 and GPIO 3 pins in BOLD along with the function names in parentheses (I2C1 SDA) 
and (I2C1 SCL) respectively, but no mention of the "Grove" connector pin numbers! Compare it to the specification given for GPIO 12 and GPIO 13 (GP12 pin 4) and (GP12 pin 3) respectively, which are WRONG! GPIO 12 should be (GP12 pin 1) while GPIO 13 should be (GP12 pin 2).
If anybody really cares!

axelsimon, 2020-12-03:
Thanks for the report @ray-leiter. Pinout.xyz is a community service website, i think @Gadgetoid does his best to update it but i also think it's not fair to expect of him to be able to fix stuff this fast.

One option if you want to make things faster it to submit a pull request with the patch needed to fix the errors. I'm currently looking at how the site is built to be able to do that, but maybe you'll figure it out faster than i can!

ray-leiter, 2020-12-03:
Thanks for the reply.
I only sent the ‘nasty’ to see if anyone was listening – never really expected a quick fix.
I’ll just wait until someone can get to it.
I don’t really want to learn much about GitHub – it’s not affecting my efforts.

Thanks again;
Ray

Gadgetoid, 2020-12-03:
I care.  But I'm also busy juggling multiple day-job projects, a house move gone awry, a toddler and - uh - what's that thing ransacking the whole world again? Ah yes. The pandemic.

I appreciate Pinout is a very popular/visible website and errors are a Bad Thing, but realistically I'm not going to get around to this quickly!

closed 2026-08-20 by Gadgetoid