pinout-xyz/Pinout.xyz #392: Add Anybeam projector from Nebra / Pi Supply

axelsimon, 2020-12-03:
Looking for pinout info, to see if it is compatible with the Fan SHIM, i discovered the Anybeam projector HAT is not on Pinout.xyz.
Shock, horror.

If i can find the pinout, i'll share it here, but opening this issue in the meantime as a reminder (to myself) / invitation to others to help find them.

https://uk.pi-supply.com/products/nebra-anybeam-hat-laser-projector-for-the-raspberry-pi
https://learn.pi-supply.com/make/getting-started-with-the-nebra-anybeam-hat/

Gadgetoid, 2020-12-03:
Subtle hint mode engaged:

👉 👉 👉 👉 👉 👉 👉 @ryanteck 👈 👈 👈 👈 👈 👈 👈 

Subtle hint mode disenagaged.

ryanteck, 2020-12-03:
I've got this over at https://github.com/NebraLtd/AnyBeam/issues/9 already!

I'll try and get this done shortly.

Basically it's every single pin because it's DPI.

axelsimon, 2020-12-03:
Sounds like i should not try to power it on with my Fan SHIM then.
Hmm, sadly the metal heatsink case isn't compatible with the projector HAT either, so not sure i can cool the rpi4 with the Nebra Anybeam HAT on then.

axelsimon, 2021-12-04:
It's been a year without any movement. Could we confirm the Nebra Anybeam is an abandoned product and then close this issue?

closed 2026-08-18 by Gadgetoid