pinout-xyz/Pinout.xyz #399: Add Raspberry Pi Pico Support

OskarZyg, 2021-01-24:


red-scorp, 2021-01-29:
Pi PIco has also a completely different pinout as bigger Pis. Try to run linux there for starters.
Pi Pico is absolutely different segment.

Gadgetoid, 2021-01-29:
I’m considering it. 🤣

But-
1. It doesn’t have the same mess of pin-exclusive functions as the Pi
2. Combinations of boards or hardware would be less likely
3. The “Pico” doesn’t really define a form-factor like “HAT” does- it will be eclipsed by boards implementing the RP2040 chip that aren’t necessarily pin-compatible

It might be that I’ll have to add “pico/RP2040” as a general site and come up with some way to represent them with the same interactive, informational style as pinout. I’m thinking SVG illustrations of the board with CSS-driven interactivity. I’d like to avoid or even eliminate JavaScript.

I also have approximately 0 time to allocate to this project since today we get the keys for our new home 😬

lurch, 2021-01-29:
4. There's already a fairly decent pinout diagram at https://www.raspberrypi.org/documentation/pico/getting-started/ :wink: 

Good luck with your moving @Gadgetoid ! :house: :crossed_fingers:

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for raspberry pi pico projects.

OskarZyg, 2021-02-13:
Advertising on Github issues?

Gordontgopher, 2021-04-14:
You may want to tell Pimoroni that the site doesn't support Pico :) (see link at bottom of [https://shop.pimoroni.com/products/pico-omnibus](url) as an example.)

Gadgetoid, 2021-05-26:
I will... :eyes: totally tell those... peemo ronnie folks... what fools! :eyes:  yes.

seamusdemora, 2022-11-15:
> Just a start - https://pico.pinout.xyz/

Great start! I'd offer only one suggestion: Make this easier to find - maybe a link from the main/home page. I visited your site looking for it, thought it odd that there wasn't a "pico page", and later "got lucky" reading this issue.

OskarZyg, 2022-11-15:
> Just a start - https://pico.pinout.xyz/

Looks amazing - very useful, some things like readability and layout are hard on mobile, especially due to little contrast in foreground and background for some text.

kllsamui, 2023-02-03:
i like it, but i fail to find the link to it from

Gadgetoid, 2023-02-03:
> Looks amazing - very useful, some things like readability and layout are hard on mobile, especially due to little contrast in foreground and background for some text.

I got told off by Google's PageSpeed for this. I have some ideas to try and fix it. Along these lines:

[image]

lurch, 2023-05-24:
@Gadgetoid Is this issue "done" now?

Gadgetoid, 2023-05-25:
I suppose it is! Thanks for the nudge.

closed 2023-05-25 by Gadgetoid