pinout-xyz/Pinout.xyz #407: I2S not I2C

urfdvw, 2021-02-27:
For the DAC boards, the DAC interface is I2S, not I2C. Please fix the description sections.

Gadgetoid, 2021-04-27:
I'm not sure if this was intended as an issue or some kind of challenge, but for future reference linking some examples sure would be helpful.

If you mean what I think you mean - this is intentional. The boards you mention *do* communicate over i2c.

The "Details" sections are auto-generated based upon the usage/mode of pins, and perhaps there's a case to add I2S here, but the I2C entries are not a mistake.

urfdvw, 2021-04-27:
As is pointed out in the diagram on page https://pinout.xyz/pinout/pirate_audio_headphone_amp.

I2C pins GPIO2 and 3 are NOT used. But I2S pins GPIO 18, 19, 21 are in use.

The same info is given on Pirate Audio's page here: https://shop.pimoroni.com/products/pirate-audio-headphone-amp
[image]

The same issue also appeared on:
https://pinout.xyz/pinout/pirate_audio_3w_amp
https://pinout.xyz/pinout/pirate_audio_line_out
https://pinout.xyz/pinout/pirate_audio_speaker

Why do I need to challenge you? I am a user of the board, I am here to offer help to improve the accuracy.

lurch, 2021-04-27:
> As is pointed out in the diagram on page https://pinout.xyz/pinout/pirate_audio_headphone_amp

Yup, looks like e.g. https://github.com/Gadgetoid/Pinout.xyz/blob/master/src/en/overlay/pimoroni-pirate-audio-headphone-amp.md should say "I2S DAC" rather than "I2C DAC".

BTW there appears to be another "bug" on https://pinout.xyz/pinout/pirate_audio_headphone_amp too - the pinout diagram on the left shows "GPIO 24 (Button Y)" and "GPIO 20 (Button Y (Old))", but the text on the right says "On older boards you might find button Y on GPIO 20. On newer boards this has been moved to GPIO 18 to free up PCM Data-in." ? :confused: 
Maybe somebody mixed up their pin-numbers and GPIO-numbers in https://github.com/Gadgetoid/Pinout.xyz/commit/88c393f2fedcbc394eb05d09c661bea230af2193 :wink:

Gadgetoid, 2021-04-27:
fuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuu

Gadgetoid, 2021-04-27:
i r a profesionul
[image]

Gadgetoid, 2021-04-27:
> Maybe somebody mixed up their pin-numbers and GPIO-numbers in [88c393f](https://github.com/Gadgetoid/Pinout.xyz/commit/88c393f2fedcbc394eb05d09c661bea230af2193) wink

ugh I get as good as I give, huh

thanks @urfdvw you're right and I'm patching this now and will conduct self flagellation later by hitting the back of my throat with hot whiskey

lurch, 2021-04-27:
> fuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuu

Sounds like you need a https://pinout.xyz/pinout/four_letter_phat :stuck_out_tongue_winking_eye:

Gadgetoid, 2021-05-26:
Right I think this fix is published now! Thank you all

closed 2021-05-26 by Gadgetoid