pinout-xyz/Pinout.xyz #415: Add Pimoronni Audio Amp SHIM and Audio DAC SHIM

Gadgetoid, 2021-05-12:


lurch, 2021-05-12:
Which one is the PIrate and which one is the NInja?

Gadgetoid, 2021-05-20:
I don't know, but I had totally forgotten this, opened this issue and gave myself a laugh. 😆 

We should prop up a local bar sometime, do our bit for the economy, y'know?

lurch, 2021-05-20:
Fo shure! Lemme know the next time you're down this way :beer:  (although we might only be allowed to prop up a table, rather than propping up the bar!)

Gadgetoid, 2021-05-20:
I live down this way now 😆

lurch, 2021-05-20:
[image]

Gadgetoid, 2021-05-20:
Well that's just a bare-faced lie! I should probably update it 😆  *updates quickly*

lurch, 2021-05-20:
[image]

[image]

Snap! :grinning:

Gadgetoid, 2021-05-20:
Although it's a bloody stretch to call here "Cambridge"! 👀  It's close enough.

closed 2021-05-26 by Gadgetoid