pinout-xyz/Pinout.xyz #421: Add OpenOCD's pinout to JTAG info page

PoroCYon, 2021-06-01:
OpenOCD uses a different pinout than the standard Alt4/Alt5 ones for JTAG, see [line 21 in this file](https://repo.or.cz/openocd.git/blob/HEAD:/tcl/interface/raspberrypi-native.cfg#l21).

lurch, 2021-06-01:
Is that OpenOCD code used for talking **from** the RaspberryPi **to** another device? (if that _is_ the case, I suspect they could be whichever pins you choose?)
Whereas I believe the "JTAG" pins displayed by pinout.xyz are those that you would use if you were talking **to** the RaspberryPi, e.g. if you were debugging bare-metal code?

PoroCYon, 2021-06-02:
I think that is the case, yes. The direction of the JTAG pins not really being indicated on pinout.xyz is what had me confused & use the wrong pins when trying to use my Pi as a JTAG probe... Maybe at least a note in the description for JTAG could be added.

Gadgetoid, 2026-08-20:
Updated the page, day late and a dollar short but feedback would be welcome!

closed 2026-08-20 by Gadgetoid

PoroCYon, 2026-08-25:
This looks good, thanks a lot!