pinout-xyz/Pinout.xyz #424: Add pinout for Raspberry Pi 1 (26-pin header)

rgm3, 2021-07-12:
I recently resurrected some of my old Raspberry Pis and couldn't find a way to get the web interface to show me the "classic" 26-pin header, although `pinout` on the command line works well.  It would be nice, for completeness, to offer the 26 pin pinout via the website.  It seems like hardware pins 3, 5, and 13 have different GPIO numbers on the old boards.

lurch, 2021-07-12:
See also #346

rgm3, 2021-07-12:
I found this very helpful site that informed me that there are several different 26-pin header configurations for revisions 0002 - 000f.
http://www.gejanssen.com/howto/hardware-revisions/index.html

lurch, 2021-07-13:
> there are several different 26-pin header configurations

There are 2 different 26-pin header configurations :wink:  The pins on that site that are marked "DNC" were basically "don't use these pins yet, because we might change them in future", but they never did change, so the "DNC" was retroactively removed.
https://www.raspberrypi.org/blog/upcoming-board-revision/ and https://www.raspberrypi.org/blog/model-b-revision-2-0-schematics/

See https://elinux.org/RPi_BCM2835_GPIOs for a fairly comprehensive reference.

lurch, 2021-07-13:
> although pinout on the command line works well

Oh, and I forgot to mention that the [`pinout` tool](https://gpiozero.readthedocs.io/en/stable/cli_tools.html#pinout) is provided by https://github.com/gpiozero/gpiozero and is nothing to do with https://pinout.xyz/

closed 2026-08-18 by Gadgetoid