pinout-xyz/Pinout.xyz #428: UART on Pi zero

eiffel31, 2021-09-29:
On this page https://pinout.xyz/pinout/uart#
UARTs available for Pi 2, 3, 4 are listed. Adding Pi zero could be helpful.

Thanks for this very practical website.

lurch, 2021-09-29:
The pinout on the Pi Zero is identical to the pinout on the Pi2 & Pi3 (and the Pi1 A+ / B+).

lurch, 2021-09-29:
I've also just spotted that that page lists GPIOs 0, 4, 8, 12 & 14 as "TXD" and GPIOs 1, 5, 9, 13 & 15 as "RXD"; but fails to differentiate that some are for different UARTs.

And I was confused by GPIOs 0, 1, 4, 5, 8 & 9 being listed as UART pins, when https://elinux.org/RPi_BCM2835_GPIOs claims that they aren't, until I realised that this is also including the UART pins for Pi4 https://elinux.org/RPi_BCM2711_GPIOs
(In which case, https://pinout.xyz/pinout/i2c and https://pinout.xyz/pinout/spi should probably also list the additional I2C and SPI pins available on the Pi 4? See also #322 )

closed 2026-08-18 by Gadgetoid