pinout-xyz/Pinout.xyz #432: Can't serve modules present but reported as not present

flatmax, 2021-11-03:
I've installed requirements :
```
pip install -r requirements.txt
```
Reports that all requirements are already satisfied. However when I run the serve command :
```
$ make serve
./generate-html.py en
This script requires the Markdown module
Install with: sudo pip install Markdown
make: *** [Makefile:15: html] Error 1
```
Not sure what to do !

Gadgetoid, 2021-11-03:
That's... weird. Broken Markdown package, perhaps? 

Might be worth changing these lines in `generate-html.py` :

https://github.com/Gadgetoid/Pinout.xyz/blob/b8f16b9c7305a42b77b6e90b2b25c829fd7f8e57/generate-html.py#L9-L12

To drop the try/catch and just import the module to see what the *actual* error is.

flatmax, 2021-11-03:
Ok - tried that , same problem. However now worked it out.
My system runs python2.7 by default pip was installing for python3. Updated the makefile to call python3 when executing.

Gadgetoid, 2021-11-03:
I need to switch everything over to Python 3, I think!

closed 2026-08-19 by Gadgetoid