pinout-xyz/Pinout.xyz #435: Limits legend

incorrectMethod, 2021-12-04:
I recommend making a duplicate of the legend that specifies the limits for current, voltage and the low high thresholds at each pin type. Additionally a recommendation such as 20% less than max power draw from outputs to avoid low power warnings at the embedded IC's.

This would benefit the newbies and possibly header devs to quickly recognise the suitability of the pins to their project. From personal experience, first time I used a Pi the power out pins of 5V0 gave an assumption that all GPIO pins could be used in a 5V0 mode.

Just a suggestion and thank you for your good work.

lurch, 2021-12-07:
I'm not sure if your suggestion fits within the scope of pinout.xyz or not, but it sounds like you might want to have a look at https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#gpio

Gadgetoid, 2026-08-19:
Added some notes to the new GPIO page. May or may not have been a good idea!

closed 2026-08-19 by Gadgetoid