pinout-xyz/Pinout.xyz #438: SPI1 on RazPi Zero 2 W doesn't exist

cjfman, 2022-01-09:
I don't think the Zero 2 W has a second SPI controller. I just got RaspberryPi 2 W, fresh install of Raspberry Pi OS and updated. I used [this library](https://pinout.xyz/pinout/spi) to send data over SPI. When I went to open controller 1, I got a file handle error. I looked in`/dev/` there were file handles for SPI0 device 0 and 1, but nothing for SPI1.

lurch, 2022-01-10:
The Raspberry Pi Zero 2 W uses the same SoC (internally) as the Raspberry Pi 3 https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#spi-overview

If you're not seeing the devices you expect in `/dev/` you've probably not got the right [DT overlays](https://www.raspberrypi.com/documentation/computers/configuration.html#part2) set up? See https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/README#L2621 for more info.

hotrungnhan, 2022-03-08:
@lurch  how about adding a layout specific for Zero 2 ư?

lurch, 2022-03-08:
@hotrungnhan I'm not sure what you're asking for, but https://pinout.xyz/ applies to the Pi Zero 2 W, exactly the same as it does for the Pi Zero, Pi 2B, Pi 3A+, etc. etc.

hotrungnhan, 2022-03-09:
I'm sorry for leading you misunderstand, I mean that pinout did not recognize the name of pizero2  and it show pi3 when call the command
[image]

hotrungnhan, 2022-03-09:
> @hotrungnhan I'm not sure what you're asking for, but https://pinout.xyz/ applies to the Pi Zero 2 W, exactly the same as it does for the Pi Zero, Pi 2B, Pi 3A+, etc. etc.
oh im sorry , i show the issue in wrong thread <> i mean this lib https://gpiozero.readthedocs.io/en/stable/cli_tools.html

lurch, 2022-03-09:
Ahhh! The `pinout` command has nothing to do with this repo, and actually comes from https://github.com/gpiozero/gpiozero - see https://gpiozero.readthedocs.io/en/stable/cli_tools.html#pinout

Support for the Raspberry Pi Zero 2 W _has_ been added there https://github.com/gpiozero/gpiozero/pull/1004 but unfortunately there's not been a packaged release of GPIO Zero for ages.

lurch, 2022-03-09:
There's now an issue at https://github.com/gpiozero/gpiozero/issues/1031 , if anyone else comes across the same problem.

closed 2026-08-18 by Gadgetoid