pinout-xyz/Pinout.xyz #443: max current

Bra1nsen, 2022-03-29:
Hey all,

I would like to set up a pid controlled heating wire via my raspi. but i wanted to get advice from experienced experts first - i hope i'm in the right place here.

i wanted to control a mosfet via pwm and regulate a 1 meter heating wire to 80°c. does that work with the available 5v or are there problems with the current?

lurch, 2022-03-29:
For any high-current applications, it's probably worth driving the load directly from a separate supply, like is done with motors?
https://projects.raspberrypi.org/en/projects/physical-computing/14

Bra1nsen, 2022-03-29:
okay and the battery gets charged/managed by the raspy?

lurch, 2022-03-29:
I think https://forums.raspberrypi.com/ might be what you're looking for? :wink:  Lots of helpful people there.

closed 2022-05-04 by Bra1nsen