pinout-xyz/Pinout.xyz #453: DPI 565/666 support

klem-everywhere, 2022-10-26:
I'm using a DPI 666 screen, would be nice to see what pins are left for other usages like i2c.

gpio=0-9=a2
gpio=12-17=a2
gpio=20-25=a2

lurch, 2022-10-27:
@Gadgetoid The "More Information" link on https://pinout.xyz/pinout/dpi needs updating to point to https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#parallel-display-interface-dpi

lurch, 2022-10-27:
@klem-everywhere As you can see in the table I linked to, there are several different ways of configuring the DPI 565 and 666 display modes.

klem-everywhere, 2022-10-27:
> @klem-everywhere As you can see in the table I linked to, there are several different ways of configuring the DPI 565 and 666 display modes.

That's true, it would be very helpful if there's a way to show it so I could show others how I could connect a touch screen with different options. But it's true that's a very specific use case.

closed 2026-08-19 by Gadgetoid