pinout-xyz/Pinout.xyz #457: Enabling i2c0 on gpio 0 and 1

Bra1nsen, 2023-03-03:
Hey thanks for your pinout website - its amazing!!

Iam wondering, if theres a guidance for setting up the gpio pins 0 and 1 for I²C. 

[image]

Any help/hint would be appreciated.

lurch, 2023-03-04:
This is probably a question best asked at https://forums.raspberrypi.com/ ?

Bra1nsen, 2023-03-04:
Iam interested in improving the pinout webside, wouldnt it be useful for the community to integrate a linke with a guidance for that topic? Kind regards

lurch, 2023-03-05:
Ah, it wasn't clear that your initial comment was a suggestion rather than a question - apologies if my reply seems patronising in hindsight.

IMHO it's up to @Gadgetoid what he does or doesn't include on these pages :wink:

ali1234, 2023-03-05:
I would say the current text is a bit out-of-date. Hardly anything uses ID eeproms but the firmware also uses this bus to control the DSI camera. If you take it over then you lose both those things, and camera is arguably the more important one.

lurch, 2023-03-05:
> Hardly anything uses ID eeproms

Not sure that's accurate? Lot's of [HATs](https://www.raspberrypi.com/news/introducing-raspberry-pi-hats/) use them. See https://pinout.xyz/boards# for some examples (although I do acknowledge that some add-on boards don't use the ID EEPROM).

P.S. CSI = Camera, DSI = Display :wink: 

P.P.S. For more info about `config.txt` see https://www.raspberrypi.com/documentation/computers/config_txt.html and for more info about enabling i2c1 see https://www.raspberrypi.com/documentation/computers/configuration.html#part3.3

ali1234, 2023-03-05:
I count 65 out of 205 unique boards that are marked as having an ID eeprom. Not all of these will have a corresponding device tree overlay for the firmware to load.

ali1234, 2023-03-05:
There are also a couple missing such as TV HAT which I am fairly sure has an eeprom (I think all the RPiF HATs do). That seems to have been incorrectly marked as using the VC I2C pins instead.

In any case I recommend against specifically giving the instructions to take over this bus. Better to link to an explanation of why you shouldn't.

lurch, 2023-03-05:
@Bra1nsen To get a bit more background-context here - do you actually **need** to use I2C0 on the HAT-ID pins, or were you just asking out of curiosity?

Bra1nsen, 2023-03-05:
Very interesting conversion, especially the part with the csi part & thanks for the documentation link! some context: 

[image]

Iam working on computer controlled cameras for sky observation, a new fan occupies the regular i2c pins, so I was considering using the alternate pins. But now it seems to me to be a bit[ cumbersome](https://forums.raspberrypi.com/viewtopic.php?t=304975)? Luckily there is still SPI for the temperaturesensor (bme280).

lurch, 2023-03-05:
I2C allows you to have many devices connected to the same bus, provided that they each have a different address. (but that's something that I've never done myself)

Bra1nsen, 2023-03-05:
yea true, I already did that with an i2c hub breakout :) but yea, the pins arent available so I why was interested in using the alternative gpio ones..

closed 2023-03-05 by Bra1nsen

lurch, 2023-03-05:
Based on the position of the Ethernet port in that photo, that looks like a Raspberry Pi 4, and those actually have many more I2C buses than earlier models https://elinux.org/RPi_BCM2711_GPIOs (which unfortunately pinout.xyz doesn't yet display #322 ).

Bra1nsen, 2023-03-14:
thanks alot for that link lurch! 

in my case it would be nice to use gpio 6 and gpio 7
[image]

Can you explain why there are multiple gpios listed in one field? Would it also be possible to use gpio 8 and gpio 9 for example? 

Moreover do I have to change some internal pi settings for using these gpios or will it be sufficient telling the python script which gpios to use... anyway I will try myself. But yea you are a great help, thanks man!

lurch, 2023-03-14:
> Can you explain why there are multiple gpios listed in one field?

Haha, I've actually edited that page myself [in the past](https://elinux.org/index.php?title=RPi_BCM2711_GPIOs&action=history), so yes I **can** explain why there are multiple GPIOs listed in one field... :rofl: 
If you look at the table mapping GPIOs to available functions:
[image]
you can see that e.g. GPIO6 (alt-mode 5) is SDA4 and GPIO8 (alt-mode 5) is also SDA4, so that's why the second table says that SDA4 is available on both GPIO6 and GPIO8.

> Would it also be possible to use gpio 8 and gpio 9 for example?

Yes.

> Moreover do I have to change some internal pi settings for using these gpios

It's not something I've ever tried myself, but you _might_ need to change some [dtoverlay](https://www.raspberrypi.com/documentation/computers/configuration.html#part3.2) settings in [`config.txt`](https://www.raspberrypi.com/documentation/computers/config_txt.html) - see [here](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/README) for a complete list, and I guess https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/README#L2167 is the option you're looking for? 
I believe you can also use https://github.com/RPi-Distro/raspi-gpio to see what "mode" your GPIO pins have been set to.

Bra1nsen, 2023-03-21:
your help was absolutely professional. thanks for that!

```
sudo nano /boot/config.txt
dtoverlay=i2c-gpio,bus=3,i2c_gpio_sda=6,i2c_gpio_scl=7
sudo reboot
sudo i2cdetect -y 3
```
[image]