pinout-xyz/Pinout.xyz #465: Libre Computer SBCs?

virtadpt, 2023-05-24:
Do you have any plans to incorporate the pinouts for Libre Computer's SBCs, such as the [AML-S905X-CC](https://libre.computer/products/aml-s905x-cc/)?

lurch, 2023-05-24:
See also #455

Gadgetoid, 2023-05-25:
So far the only SBC manufacturer to give me the time of day has been Raxda via OKDO, who graciously supplied me a Rock 5 which I've absolutely failed to do anything with.

The trouble with anything but Pi is:
1. There are too many of them
2. Like, waaay too many, even single manufacturers churn out multiple board in different form-factors with multiple pinouts and SOCs etc- Pi's slow burn and samey products might seem frustrating, but it's *super important* for ecosystem
3. I just don't have any free time anymore
4. The ROI on Pinout is negligible, and while I don't do this for the money it's certainly a motivating factor
5. The manufacturers just don't seem to care about software/support beyond the bare minimum

Pinout as it stands is also a pretty bad, legacy mess. The idea is great, but in order to extrapolate it to any other SBCs I would need a ground-up rewrite. I think @ali1234 might have done some work to this end, but let's be real here- wading through my 10+ year old organically grown nonsense code is not a fun task.

For Pico and Pico W I did just this. Pure HTML. Ground up. And I'm pretty pleased with the results.

So yeah building a microsite for a board I don't own or care about, made by a company that doesn't know me from a stranger in the street, to satisfy their three users, only for it to be obsolete in three months... isn't a super motivating challenge :laughing: 

If there's ever a board that strikes me as super interesting and popular with a motivated push behind it- I think OKDO/Raxda are the most notable I've seen for a while- and without too much churn (companies like UDOO have always looked cool but their lineup looks like the result of a random board layout generator) then I'd reconsider.

virtadpt, 2023-05-25:
All of those reasons make perfect sense.  Thank you very much.

Also, the idea of a random board layout generator is kind of funny - it sounds like part of a hardware hacking CTF.  :)

closed 2023-05-25 by virtadpt