pinout-xyz/Pinout.xyz #466: Suggestion: Add Dark Mode

SX-9, 2023-05-27:
The current theme is too bright

Gadgetoid, 2023-05-28:
I did this for https://pico.pinout.xyz, so yeah - hard agree :laughing:

SX-9, 2023-06-04:
> I did this for https://pico.pinout.xyz, so yeah - hard agree :laughing: 
> 

do it for this site too please

Gadgetoid, 2023-06-12:
Made a start. It's not great, but it's a start!

[image]

SX-9, 2023-06-14:
> Made a start. It's not great, but it's a start!
> 
> [image]
> 
> 

what css framework are you using? i might beable to help

SX-9, 2023-06-14:
also add a theme toggle button

Gadgetoid, 2024-06-16:
This will be an explicit goal of Pinout 2024 - https://github.com/pinout-xyz/pinout-2024/issues/1

Gadgetoid, 2025-12-05:
It's 2025 going on 2026 and my grand plans for Pinout 2024 never saw fruition.

I have taken to applying some incremental improvements to Pinout as-is instead, including mobile responsiveness and - yes - now a (not so severe) dark mode.

closed 2025-12-05 by Gadgetoid