pinout-xyz/Pinout.xyz #468: RPi Zero 2W showing as "Pi Model ??? v1.0"

DFC72, 2023-07-18:
also shows incorrect info for # of usb ports, and presence of ethernet port, wifi and bluetooth.
OS is Rasbian 32-bit lite if that matters

pi@raspberrypi:~ $ pinout
,--------------------------------.
| oooooooooooooooooooo J8     +====
| 1ooooooooooooooooooo        | USB
|                             +====
|      Pi Model ??? V1.0         |
|      +----+                 +====
| |D|  |SoC |                 | USB
| |S|  |    |                 +====
| |I|  +----+                    |
|                   |C|     +======
|                   |S|     |   Net
| pwr        |HDMI| |I||A|  +======
`-| |--------|    |----|V|-------'

Revision           : 902120
SoC                : BCM2837
RAM                : 512MB
Storage            : MicroSD
USB ports          : 4 (of which 0 USB3)
Ethernet ports     : 1 (0Mbps max. speed)
Wi-fi              : False
Bluetooth          : False
Camera ports (CSI) : 1
Display ports (DSI): 1

J8:
   3V3  (1) (2)  5V
 GPIO2  (3) (4)  5V
 GPIO3  (5) (6)  GND
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8
   GND (25) (26) GPIO7
 GPIO0 (27) (28) GPIO1
 GPIO5 (29) (30) GND
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21

For further information, please refer to https://pinout.xyz/
pi@raspberrypi:~ $

lurch, 2023-07-19:
See https://github.com/gpiozero/gpiozero/issues/1037 - this isn't a bug with pinout.xyz :wink:

DFC72, 2023-07-19:
Not sure I understand? Is there more than one pinout project?

Gadgetoid, 2023-07-19:
I, uh, guess the "pinout.xyz" link in there is probably a little misleading at times :grimacing: But, no, we are indeed not affiliated in any way other than the respective project authors kinda knowing each other sorta.

closed 2023-10-13 by Gadgetoid