pinout-xyz/Pinout.xyz #470: Add suppor to raspberry pi 5

patrickelectric, 2023-09-28:
https://www.raspberrypi.com/news/introducing-raspberry-pi-5/

https://www.youtube.com/watch?v=nBtOEmUqASQ

Gadgetoid, 2023-09-28:
I have one, but I'm not yet sure how it differs and what the impact will be. I'm currently mired in trying to marshal some sensible, working solution to GPIO in Python.

patrickelectric, 2023-09-28:
Thanks @Gadgetoid!

lurch, 2023-09-28:
#322 :see_no_evil:  :rofl:

Gadgetoid, 2024-06-16:
Actual proper documentation of Pi 5 is an explicit goal of https://github.com/pinout-xyz/pinout-2024/issues/1

It deeply disturbs me that other sites with *terrible* pinout diagrams rank higher on Google just because they mention "Raspberry Pi 5 Pinout" 💀

patrickelectric, 2024-06-16:
@Gadgetoid I can help on that, just need to point how it should be done. I have a pi5 and I did work on gpio implementation for ardupilot.

Gadgetoid, 2024-06-16:
Thank you!

See: https://github.com/pinout-xyz/pinout-2024/blob/main/boards/raspberry-pi-5.json

Which intends to be a complete and concise "database" of the Pi 5's headers, pins and alternate functions.

Would be super helpful to know:

* What I'm missing
* What could be improved

Gadgetoid, 2026-08-19:
Sort of done here.

I abandoned pinout-2024, it was a distraction and it was never going to be better than improving this site.

closed 2026-08-19 by Gadgetoid