pinout-xyz/Pinout.xyz #471: Pinout shows ??? on graphic display for Pi Zero 2 W?

Jibun-no-Kage, 2023-10-13:
Pinout shows ??? on graphic display for Pi Zero 2 W?  But /proc/cpuinfo shows correct information.

,--------------------------------.
| oooooooooooooooooooo J8     +====
| 1ooooooooooooooooooo        | USB
|                             +====
|      Pi Model ??? V1.0         |
|      +----+                 +====
| |D|  |SoC |                 | USB
| |S|  |    |                 +====
| |I|  +----+                    |
|                   |C|     +======
|                   |S|     |   Net
| pwr        |HDMI| |I||A|  +======
`-| |--------|    |----|V|-------'

Revision           : 902120
SoC                : BCM2837
RAM                : 512MB
Storage            : MicroSD
USB ports          : 4 (of which 0 USB3)
Ethernet ports     : 1 (0Mbps max. speed)
Wi-fi              : False
Bluetooth          : False
Camera ports (CSI) : 1
Display ports (DSI): 1

Why does it also show WiFi not present?  Symptom of the same issue?

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 000000008f249728
Model           : Raspberry Pi Zero 2 W Rev 1.0

Gadgetoid, 2023-10-13:
This repository is unrelated to gpiozero's "pinout" tool. Try here: https://github.com/gpiozero/gpiozero

closed 2023-10-13 by Gadgetoid

Jibun-no-Kage, 2023-10-13:
Ah... thanks.