pinout-xyz/Pinout.xyz #472: Missing non-English subdomains

Gadgetoid, 2023-10-13:
I had to do an emergency migration from AWS to GitHub pages a while back, and decided to stick with pages since it drastically simplifies my stack and workflow.

However, all of the non-English subdomains have been broken.

I'm not really sure how to fix this in GitHub pages, but I think in practice the only solution is to have individual GitHub (pages content only) repositories for each language which can be populated by the CI in #462 ... somehow.

lurch, 2023-10-13:
Possibly via some cunning usage of https://docs.github.com/en/actions/security-guides/using-secrets-in-github-actions ?

ali1234, 2023-10-14:
I looked in to this in #448. If you want subdomains on GH Pages then you do have to have a repository for each one.

I would instead recommend refactoring the whole site so everything is served from subdirectories instead. Then set up the subdomains to send 301 Permanently Moved. That will need to be hosted off github. DNS providers usually have a way to set it up for free.

Gadgetoid, 2024-02-12:
Subdirectories makes a lot of sense, I'll have to look into that. Would definitely prefer the whole thing to be automatic and uncomplicated.

Gadgetoid, 2024-06-16:
As per https://github.com/pinout-xyz/pinout-2024/issues/1 and unless we figure out otherwise, translations will not be a going concern of Pinout in future.

closed 2024-06-16 by Gadgetoid