pinout-xyz/Pinout.xyz #474: feature: show release date and manufacturer based on board revision

roycewilliams, 2023-10-23:
It would be useful to extend lookup tables to map revision ID to the 'Release Date' and manufacturer (as shown in the Notes field here). This would help to determine board age and origin.

I'm not sure which lists are most definitive, but this one looks solid:

https://elinux.org/RPi_HardwareHistory#Board_Revision_History

lurch, 2023-10-23:
I think you might be confusing the https://pinout.xyz website with the [`pinout` tool](https://gpiozero.readthedocs.io/en/latest/cli_pinout.html) provided by [GpioZero](https://github.com/gpiozero/gpiozero/) ?

roycewilliams, 2023-10-23:
Ah, well ... when a naive user (such as myself ;) runs `pinout`, which comes with Raspberry Pi OS by default, the last line of output is:

`For further information, please refer to https://pinout.xyz/`

If that's not the way for people to discover how to interact with the project, that's unfortunate. Whichever team is responsible for the output above should probably do something to disambiguate. :D

Gadgetoid, 2023-10-23:
Yeah that link is *confusing* to say the very least. As nice as it is to have a semi-official tool acknowledge this unofficial project 😆 

Uh @bennuttall I think that was your doing? Any thoughts... we could stick an interstitial page in the flow, change it to something like "For an unofficial Pinout diagram, refer to", or just drop the link in favour of something more official?

bennuttall, 2023-10-23:
This was just an attempt to avoid adding all the SPI/I2C/etc info to the tool, as that's covered in INTERACTIVE GOODNESS over at .xyz

Extra fun: run `pinout -x` to open THE OFFICIAL WEBSITE OF PINS in your browser: https://gpiozero.readthedocs.io/en/latest/cli_pinout.html#cmdoption-pinout-x

It didn't occur to me that people would run `pinout` not knowing it came from gpiozero. Maybe we could add that to the tool. I still like to defer to .xyz for more information than I want to provide.

roycewilliams, 2023-10-23:
Moved to https://github.com/gpiozero/gpiozero/issues/1097 .

closed 2023-10-23 by roycewilliams