pinout-xyz/Pinout.xyz #475: BeagleBoard.org SBC pinout

lorforlinux, 2023-11-20:
Hey @Gadgetoid, thanks for making this wonderful interactive pinout website. Will it be possible to create a site like `beagleboard.pinout.xyz` to host the pinout diagram of [fully OpenSource SBC ](https://www.beagleboard.org/boards) from BeagleBoard.org?

anujdeshpande, 2023-11-20:
Happy to pick this up!

@Gadgetoid how about we do a rear view like system for the 2 headers - similar to what you have on the pico.pinout page?

So there's 2 main headers - P8 and P9 as they are commonly called - that need to be documented for the BeagleBone series of boards.

It will look something like this - 
[image]

cc @lorforlinux

Gadgetoid, 2023-11-20:
Happy to make this a “.pinout.xyz” project if you want me to fire up a GitHub repo and a subdomain. I can help with the HTML side and some rudimentary accessibility knowledge, but without hardware I can’t do anything fancy. Who knows, maybe we’ll get some upstream support 🤣

For complex SBCs like this, it would be nice to have recipes for enabling/configuring interfaces. The Pi pinout documents some GPIO programming stuff but it’s mostly quite thoroughly obsolete now.

Gadgetoid, 2023-11-20:
(Didn’t mean to close, mashed the button on my phone!)

lorforlinux, 2023-11-27:
@Gadgetoid @anujdeshpande I think what we have here is easiest to port -> https://docs.beagleboard.io/latest/boards/beaglev/fire/01-introduction.html#pinout-diagrams

Grippy98, 2024-04-16:
@Gadgetoid We do have a Pi compatible board being released called BeagleY-AI, that's going to be the easiest to port since it's HAT compatible and mostly just adding some of the pinmux features vs what's already there for Pi. I'll fork this current repo for now but maybe can you spin up something like beagley.pinout.xyz? 

Eventually it would be cool to have just a beagle.pinout.xyz to be a bit more in-depth and include all the BeagleBone form factor boards.

Gadgetoid, 2024-06-13:
@Grippy98 you’re aware the main thrust of project is CC *NC* *BY*, right?

In no way did the discussion here endorse “strip the website of all credit and slap it up on another URL or promote sales of our new SBC”. 😣

Gadgetoid, 2024-06-13:
Suffice to say I’m not happy. Freebooting the entire website for a commercial SBC is bad enough without scrubbing any reference to me or its origins.

But you know what really twists the knife… removing *just* my name from the credits in the source, too 😐

[image]

This website might be the first experience many users get with a body of work I’ve been maintaining for a decade and you’ve made real actual effort to make sure it *doesn’t* contain attribution. Smdh.

lorforlinux, 2024-06-13:
Hey @Gadgetoid, I understand this is not good in any way and we are very sorry to make you feel bad. I hope we can resolve this issue, this is a mistake from our end and we will do our best to resolve this.

> But you know what really twists the knife… removing just my name from the credits in the source, too 😐

I think @Grippy98 just wanted to add a Beagleboard reference but accidentally updated over your name because the link is still pointing to your GitHub username. 

[image]

I see that the references are also stripped from the footer of the main site which makes this situation worse. I will get those added back to the site today.

> In no way did the discussion here endorse “strip the website of all credit and slap it up on another URL or promote sales of our new SBC”. 😣

That's understandable and we are open to knowing how to get this sorted ASAP. BeagleBoard.org is a not-for-profit organization, we don't want to make this mistake define how we work. I started this discussion hoping we could collaborate and create a subdomain, I hope you understand what just happened was not our intention.

> Suffice to say I’m not happy. Freebooting the entire website for a commercial SBC is bad enough without scrubbing any reference to me or its origins.

My apologies for making you unhappy, you have invested your heart and soul into this project. I will make sure all the references and attributions will be added back to our forked repository. If you have anything else in mind, please let me know. I am hoping we can resolve our mistake with peace, I am sorry to hurt you.

Gadgetoid, 2024-06-13:
There’s not really a “whoops sorry” out of such a blatant scrubbing and freebooting of a website, your organisation should not allow systemic failures like this- *multiple* people were involved in discussing, making, reviewing and merging these changes -, and the individuals within it should certainly know better.

ICYMI from Discord:

> I can’t believe I’m having to say these words to BeagleBoard of all institutions- but can whoever is responsible for https://pinout.beagley.ai take it down immediately, please?

> If you’re in a desperate hurry for OpenSauce marketing materials then I am available to discuss terms for a licensing exemption.

I’m sorry, but there are no modifications you can make which would make the website compliant with our license.

Until you remove the offending material we have no amicable grounds for discussion.

You are not a “not for profit” organisation in any meaningful sense of the term. You exist primarily to promote TI products and all of the veneer of FOSS you apply over this goal does not make me sympathetic to your cause.

Andrei’s job title is *marketing* at TI. Smdh. Don’t insult my intelligence.

Grippy98, 2024-06-13:
@Gadgetoid I'm just about to leave to drive to the airport right now but as soon as I get there I'll wipe the site until we can talk. 

I'm genuinely really sorry, I did realize this week that at some point I have removed the main links that point back to you and Pinot and I can literally show you my "notion" to-do tasks to add them back. I don't really remember why/how I took them down but please understand it's completely a misunderstanding. When I first forked the project, I did change the home page but I had plenty of "If you enjoy this, please credit the original creator, references both on the front page and as a second line on the Footer"

You have every right to be angry and I personally apologize, please understand that I didn't do it out of malice in any way and the intention on our end was always to work together on this if we could. I genuinely enjoy Pinout and I think you did an amazing job creating and maintaining it, I'm genuinely a fan. 

As for my role at TI, yes I realize how it looks but also please understand that I spend more time on Beagle as my own hobby than anything, in many ways it's a passion project for me.

Let me know if you're available to talk, my personal email is on GitHub and as I said I'll be at opensauce so if I can apologize in person I would like to as well. 

I'm blanking out the content of pinout.beagleboard.org until I have your approval to put anything back.

Gadgetoid, 2024-06-13:
@Grippy98 thank you. 

Judging by the repository history, it looks like Open Sauce had put a little time pressure on you, which- to be fair- I can empathize with.

This doesn't really explain how a non-commercial project was forked to promote/document a very plainly commercial product, though. Regardless of the roles, incentives and actions of individuals involved that's a move I wouldn't - if I were BeagleBoard dot org - make lightly.

That said - I don't believe for a moment there was any malice involved and I understand these things can be messy and complicated and sometimes our enthusiasm outpaces our rationality.

Regrettably I will not be attending Open Source, but enjoy your trip and with any luck we can work something out.

For the record I am not some master tactician who timed my outrage to coincide with the event. I really did learn about this from an Open Source exhibitor this very morning and I barely know what day it is, much less when or where events are happening.

I am actually a little disappointed that my work wont be part of the marketing thrust of a BeagleBoard product (I'm a fan from my brush with Beaglebone Black many years ago), but if I don't defend the letter of this project's license then it'll be open season on 11 years of on-and-off effort - https://web.archive.org/web/20130505194305/pi.gadgetoid.com/pinout

The NC clause is a necessary evil. Attribution and footers are nice 'n' all, but I can't buy my kids LEGO with words. It's a shame that - afacit - your love of this project didn't run deep enough for you to click one of my sponsorship links before deleting them :laughing:  :skull:

Grippy98, 2024-06-13:
Thank you as well I appreciate your understanding... really has been a combo of self-inflicted pressure with the show and excitement to get it out there. I have no thought whatsoever that you tried to time this or anything like that and of course you should defend your project. 

And you're right I should have haha, I don't make the big bux either but consider that "LEGO set" my personal apology. :) I should have known/done better. A lesson well learned on my part to not be careless with these things.

Gadgetoid, 2024-06-13:
> I don't make the big bux either

Yeah this whole ecosystem is definitely more hobby with a salary than those get-paid-seven-figures-to-keep-a-chair-warm FAANG roles :laughing: :sob: 

Say hi to Paul at OpenSource if you cross paths, and also Chris who will be there with Long Pong, in which I had a vanishingly small amount of involvement and he just absolutely ran with.

Grippy98, 2024-06-13:
When they said "you should learn python" they meant for data science not microcontrollers doh! 😛 

Will do! Long Pong looks crazy haha, can't wait to see it

Gadgetoid, 2024-06-13:
I guess this really shows the place I was at mentally in 2017 vs now huh - https://github.com/pinout-xyz/Pinout.xyz/issues/218

That issue led to this website here- https://www.96boards.org/pinout/

Which has remained 7 years old and kinda ugly forever, apparently. *sigh*

Gadgetoid, 2024-06-16:
After all this kerfuffle I have been nerd sniped into finally starting a project to reconsider what Pinout is and how it should be generated.

If you're interested in being part of the conversation and/or effort making Pinout's (hopefully inevitable) replacement, see: https://github.com/pinout-xyz/pinout-2024/issues/1

closed 2026-08-19 by Gadgetoid