pinout-xyz/Pinout.xyz #480: Discussion on adding pins for non-Raspberry Pi products

armsom-team, 2024-06-12:
Thank you very much, we noticed that Pinout is very carefully done. We would like to know if it is possible to submit the pin definition of armsom-sige7 on your website [https://docs.armsom.org/armsom-sige7#hardware-pin-definitions](https://docs.armsom.org/armsom-sige7#hardware-pin-definitions).
If possible, we are willing to submit it and make some contribution to the website.

Gadgetoid, 2024-06-13:
Thank you for your interest.

I don't really have an answer to this question yet. While I always wanted Pinout to be a catch-all for decent, accessible GPIO pinout documents... I don't really have the time or inclination to maintain documentation for an SBC that I don't use on a regular basis.

As it stands, there are many improvements I would like the time and inclination to make on Pinout itself.

I am also torn between my FOSS ideals, my desire for better documentation out there, and my not wanting commercial hardware manufacturers to have a free lunch at my (and the other maintainers here) expense.

This is not a no, but it's not a yes either. I need to figure out what I want from this project, what direction it should take, and how I could accommodate commercial interests.

To even make maintaining multiple boards feasible under a single umbrella - without a scatter of disparate, out of sync GitHub repositories - would be a significant undertaking.

In the near term, see https://github.com/pinout-xyz/Pinout.xyz/issues/475 for an example of what *not* to do.

Gadgetoid, 2024-06-13:
I should probably also note that I *have* expressed discomfort with the Non Commercial license clause before, and *have* extended ad-hoc exemption from it. See: https://github.com/pinout-xyz/Pinout.xyz/issues/218

However, the absolute atrocity that 96boards have unleashed upon the internet makes me hesitant to do this again - https://www.96boards.org/pinout/

Gadgetoid, 2024-06-14:
TLDR, the NC clause has been dropped.

Please feel free to take Pinout and build, host your own documentation website with appropriate attribution. It publishes to GitHub pages automatically so it should probably be quite easy.

See: https://github.com/pinout-xyz/Pinout.xyz/commit/aa82f18c0e908af17dc342ce99d1abf042d4db3f

And: https://github.com/pinout-xyz/Pinout.xyz/issues/481

70k Pi users refer to Pinout.xyz every month. If you're interested in the promotional value of Pinout.xyz then I suggest you reach out to me regarding a nondescript paid header or footer link once your documentation effort is completed. I squandered all the Patreon funds (turns out they build up over 10 years, anyone can do the math) on a monitor I no longer use (it ended up kinda sucking), Steam Decks (yes, plural) and fancy key caps 🫠 and there's a bunch of stuff I still want to bring myself up to speed with the makersphere. Or at least try to.

Gadgetoid, 2024-06-16:
See also: https://github.com/pinout-xyz/pinout-2024/issues/1

closed 2026-08-19 by Gadgetoid