pinout-xyz/Pinout.xyz #481: RFC: License Change

Gadgetoid, 2024-06-13:
I’m considering relicensing this project to unambiguously permit commercial use and re-use of any kind under any circumstances.

Pinout started as a simple little hobby project under the umbrella of a small Pi blog I started back in 2013. Here it is- all new and clunky, with that 26-pin header: https://web.archive.org/web/20130505194305/pi.gadgetoid.com/pinout

I never really had ambitions for it, but it turned out to be pretty popular so it was a nice way to make a name for myself apart from my affiliations and subsequent employment with Pimoroni.

A lot has changed since those humble beginnings. I don’t do as much work with the Pi - Pico is the new hotness - I have a lot of plates spinning, I have two additional kids, and I haven’t really succeeded in finding a place for myself in the maker sphere so I’m hunting elsewhere.

As such you’ve probably noticed that Pinout isn’t well maintained. Hasn’t been updated with some of the improvements I was working on. And doesn’t even mention the Pi 5!

In truth I’m also a little weary of the project, it has mostly existed as a little marketing bandwagon for manufacturers to get eyes on their new PCBs. I’ve been happy about that, since I prefer to work with, not against, others in this space. But declining traffic and declining financial support (though I am eternally grateful to people who did support me for this and other mad ventures) hasn’t exactly reinvigorated a passion in me.

Today has pretty much buried what was left of that passion. Today this project consumed a day of my time on license busywork and FOSS politics instead of improvements and accessibility. Today this project got me up at 6am, fixated on my phone, and resulted in a falling out with my middle child. Today has not been fun.

In light of todays events and both past and present requests to take Pinout and make documentation for other SBCs, and with the realisation that I don’t have the time or energy to collate these all here under the one umbrella I’d have loved to, I’m considering relicensing Pinout.xyz to a “do what you want” License. This is with the hope that those with respect for the project will credit it, and those who stand to benefit from the project might throw me a bone. But also that maybe we all get slightly better documentation. And I will get slightly better sleep.

Above all else I’m tired of getting asked the same thing over and over again-

* https://github.com/pinout-xyz/Pinout.xyz/issues/480
* https://github.com/pinout-xyz/Pinout.xyz/issues/218
* https://github.com/pinout-xyz/Pinout.xyz/issues/475
* https://github.com/pinout-xyz/Pinout.xyz/issues/455
* https://github.com/pinout-xyz/Pinout.xyz/issues/465
 
So I out this question to the community - and especially @lurch who has been with me every step of the way and super helpful and proactive across more projects than I dare count (I owe you like ten meals and a cool hat by now) - should I YOLO, relicense and choose a quiet life?

As for the Pinout.xyz website… it’s in bad shape, I kinda want to rewrite it but I’m desperately low on time or energy. Maybe it’ll happen someday! The Pico sites are so much easier to maintain, so much less hassle, and a bit more accessible to boot. It’ll continue to live on in its present form- maybe I can scratch up a dollar or two selling too-bar links to all the clones!? 🤣

lurch, 2024-06-13:
Hey Phil, I'm sorry to hear that Pinout.xyz has given you such a rough day, and that you've lost your passion for it. Totally understandable though :hugs: 
I also have a lot less time and motivation to work on random 3rd-party FOSS projects (like Pinout.xyz and GpioZero) than I did several years ago.

I don't have any strong opinions either way, and IMHO it's entirely up to you what you do with this project going forwards. Just because the code is open-source, that doesn't mean you have to open-source your decision-making too :rofl:

Gadgetoid, 2024-06-14:
Noted, and irrevocably relicensed: https://github.com/pinout-xyz/Pinout.xyz/commit/aa82f18c0e908af17dc342ce99d1abf042d4db3f

I do prefer to open source my decision making, it feels like a lot of people have contributed a lot of effort here - you in particular - and I don't like to tread on toes or make people feel uninvolved. In this case, though, I feel I must act quickly and decisively for my own peace of mind.

Gadgetoid, 2024-06-16:
Note: since this Pinout repository represents pretty much a dead end for progress on this project, I’m also considering adding some indication of this fact and - yes I got nerd sniped again - starting a fresh project to rebuild Pinout from the ground up with consideration for alternate boards, better accessibility and a visual style fit for this century. Not to mention better tooling and static generation, sheesh it’s pretty bad!

I have a strong temptation to strip down the main page and build something that more closely reflects pico.pinout.xyz and includes more up-front information.

This doesn’t mean *dropping* HATs and pHATs but they are - at present - a bit of a mess too.

My brain won’t let go of this 🫠

lurch, 2024-06-16:
> starting a fresh project to rebuild Pinout from the ground up with consideration for alternate boards, better accessibility and a visual style fit for this century. Not to mention better tooling and static generation, sheesh it’s pretty bad!

If you're going to tackle that (sounds ambitious!) can I suggest (for ease of future maintenance!) that you make more of an effort to separate "data" from "presentation"? :slightly_smiling_face:  (I don't mean to sound rude, but that's one of the things that always niggled me about pinout.xyz - but I guess that's probably because you came at it from the HTML & CSS side of things, whereas I was always looking at it from the Python side of things.)

Also, given that the site **is** generated programmatically, it would be nice if you used a "proper" translation / localisation (l10n) framework, rather than doing "copy all these MarkDown files verbatim and just translate the bits that you think need updating" :laughing: 
a) this makes it easier for the translators to see exactly which strings need updating
b) it makes it easier for you (as the project-maintainer) to see which strings haven't been translated yet
c) it means you can change the design of the "English" version of the site, and you don't get stuck with the non-English sites still using the old design

All IMVHO of course!

> This doesn’t mean _dropping_ HATs and pHATs but they are - at present - a bit of a mess too.

Again, my programmer-brain approach would be to have these generated from e.g JSON files, rather than relying on "special comments" inside a [markdown file](https://raw.githubusercontent.com/pinout-xyz/Pinout.xyz/master/src/en/overlay/pimoroni-inky-what.md). And then if you wanted to get _really_ fancy you could validate them using https://json-schema.org/

> My brain won’t let go of this

:rofl: 

> I have a strong temptation to strip down the main page and build something that more closely reflects pico.pinout.xyz and includes more up-front information.

Given the topic of this PR, perhaps you ought to think about adding _some_ kind of licence to https://github.com/pinout-xyz/pico.pinout.xyz too? :thinking: :cold_sweat:

Gadgetoid, 2024-06-16:
> If you're going to tackle that (sounds ambitious!) can I suggest (for ease of future maintenance!) that you make more of an effort to separate "data" from "presentation"? 🙂

This is probably one of the primary goals 😆 

> Also, given that the site **is** generated programmatically, it would be nice if you used a "proper" translation / localisation (l10n) framework

I will probably not attempt to tackle translation at all. I'm told that a bad translation is *worse* than a good English language site.

Also I10n and everything to do with it is just the most joyless, arduous experience so I don't have much appetite to try. I'm inclined to just slap a permissive enough license on the project that folks native to "insert language here" can just wholesale copy it, make it their own, and make it good without needing me to accept PRs under the pretence that I can understand more than five words.

> Again, my programmer-brain approach would be to have these generated from e.g JSON files, rather than relying on "special comments" inside a [markdown [...]

Yeah we really did get quite abstract with all that. I might be inclined to pair a .json file with a .md file because Markdown is nice, and Markdown embedded in JSON is not nice.

> Given the topic of this PR, perhaps you ought to think about adding _some_ kind of licence to https://github.com/pinout-xyz/pico.pinout.xyz too? 🤔 😰

Uuuuh, fair point! I'm less inclined to license that permissively because it's already a solved problem and people can make their own damn website 😆 but maybe I'm just being precious.

Gadgetoid, 2024-06-16:
THAS ONE BIG SCHEMA MUMMA https://github.com/pinout-xyz/pinout-2024/blob/main/schema/__init__.py

(disclaimer: probably actually quite concise)

lurch, 2024-06-16:
> > If you're going to tackle that (sounds ambitious!) can I suggest (for ease of future maintenance!) that you make more of an effort to separate "data" from "presentation"? 🙂
> 
> This is probably one of the primary goals 😆

:tada: 

> I will probably not attempt to tackle translation at all. I'm told that a bad translation is _worse_ than a good English language site.

Fair enough. It'll cut down on your workload if nothing else.

[HAT definitions]
> Yeah we really did get quite abstract with all that. I might be inclined to pair a .json file with a .md file because Markdown is nice, and Markdown embedded in JSON is not nice.

Probably makes sense to have a directory-per-board containing `data.json` and `README.md` and a `board.jpg` ?

> Edit: It probably goes without saying- but it shouldn't- that I appreciate your input! Thank you.

:heart:

Gadgetoid, 2024-06-16:
> Probably makes sense to have a directory-per-board

I was thinking along those lines, yes! Also leaves room for expansion- more images or... whatever else makes sense to collate.

closed 2026-08-18 by Gadgetoid