pinout-xyz/Pinout.xyz #484: Other languages are broken and Italian is a slot machine ad

Zipdox2, 2024-11-25:
All the languages other than English give a 404, except Italian (https://it.pinout.xyz/), which appears to have been hijacked and shows a slot machine ad.

Gadgetoid, 2024-11-25:
The non-English sites got lost in the transition over to GitHub pages.

But also... WHAT!? How on earth has the it site been hijacked!? It's a CNAME to `pinout-xyz.github.io`!?

Gadgetoid, 2024-11-25:
Okay apparently if I leave a CNAME pointing to a domain pointing to GitHub pages anyone can just rock up and claim it with whatever spam website they like and I have to grapple with support to try and reclaim it. Good fun. 🫠

lurch, 2024-11-26:
> The non-English sites got lost in the transition over to GitHub pages.

Will they be returning, or does it make sense to remove the flags from the bottom of the page?

> Okay apparently if I leave a CNAME pointing to a domain pointing to GitHub pages anyone can just rock up and claim it with whatever spam website they like and I have to grapple with support to try and reclaim it. Good fun. 🫠

OOOOF! :sob:

Gadgetoid, 2024-11-26:
> Will they be returning

I wanted to try and bring them back but I've just never found the willpower or the energy.

Think the solution is to slap up a repository for every subdomain, claim the subdomain (troublesome for it.pinout.xyz because it's currently "owned" by a malicious GitHub user *sigh*), and point the publish script output at that repo somehow.

But then also I want to completely rebuild pinout from the ground up (if I can bring myself to stop hating https://pinout-xyz.github.io/pinout-2024/) sooooooooo 🫠

lurch, 2024-11-26:
Too many side-projects, not enough time or energy to devote to them - I know **exactly** what that's like! :sweat_smile:

Gadgetoid, 2026-08-18:
I ended up slicing it out into https://pi.pinout.xyz and now looking back it's bad in entirely new ways- the icons to disambiguate the colours of pins are the same kind of entirely redundant noise Apple got slammed for in macOS menus, and the Pi pinout has none of the amenities of the Pico ones.

Redoubling my efforts to improve Pinout, but bringing back translations is regrettably low on the pile 😭

Gadgetoid, 2026-08-19:
Languages have returned as sub directories. A little stale, but they're not going to get improved if they're not visible.

closed 2026-08-19 by Gadgetoid