pinout-xyz/Pinout.xyz #485: The UARTs should really have their UART number by them

factoidforrest, 2025-01-11:
The various optionally enabled uarts onboard go to specific pins. This is sort of a newer thing since the pi4. Each should have its uart number instead of just "rx" or "tx" by them in the overlay. Here's a handy table. 

| UART  | TX GPIO | RX GPIO | TX Pin | RX Pin |
|-------|---------|---------|--------|--------|
| UART0 | GPIO 14 | GPIO 15 | Pin 8  | Pin 10 |
| UART1 | GPIO 0  | GPIO 1  | Pin 27 | Pin 28 |
| UART2 | GPIO 4  | GPIO 5  | Pin 7  | Pin 29 |
| UART3 | GPIO 8  | GPIO 9  | Pin 24 | Pin 21 |
| UART4 | GPIO 12 | GPIO 13 | Pin 32 | Pin 33 |

I'd be happy to fix this if I knew a PR would be considered! Thanks

Gadgetoid, 2025-07-03:
This was something I caught doing https://pinout-xyz.github.io/pinout-2024/# so I tend to agree!

I've got a bit of a time/bandwidth problem at the moment, so this site has been neglected 😭  but a PR is provisionally welcome.

Gadgetoid, 2026-08-18:
I... think... this is fixed now!

closed 2026-08-18 by Gadgetoid