pinout-xyz/Pinout.xyz #490: Raspberry Zero W: GPIO vs. physical - what am I missing

umbertones, 2025-03-31:
Quote from the website:
"GPIO - General Purpose Input/Output, aka "BCM" or "Broadcom". These are the big numbers, e.g. "GPIO 22". You'll use these with RPi.GPIO and GPIO Zero."

my code:
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BOARD)
PIN_TRIGGER = 18
PIN_ECHO = 24

My cables where on these GPIOs from the pinout.xyz-Image....nothing worked. 

But when I use the physicals
PIN_TRIGGER = 12
PIN_ECHO = 18

Everything works as expected

I found out, when I switched to GPIO25 and got the error message:
ValueError: The channel sent is invalid on a Raspberry Pi

Any library change? Or a misunderstanding? 

Thanks in advance.

lurch, 2025-03-31:
You've done `GPIO.setmode(GPIO.BOARD)` in your code which switches RPi.GPIO from "GPIO" numbering to "physical" numbering.

umbertones, 2025-03-31:
Ah, I wasn't aware of that. Thanks for the clarification!

closed 2025-03-31 by umbertones